jorgenkraghjakobsen/snapclient #1: Installation process

dmckminard, 2020-04-23:
Hello !

This program seems pretty cool. Can you detail the installation process ? I'd like to run snapclient on an ESP32

Good job !

jorgenkraghjakobsen, 2020-04-23:
Hi Minard 
Sorry this is very much work in progress - and not working very well jet. 
WLAN UDP packges are not comming at a prediable flow - we do not have good time stamping on in comming packages. Stil a lot work in front of me. 
On the other hand if your are on wired hardware ie the wESP32 board it works. 
Regards Jorgen

dmckminard, 2020-04-23:
I'd really like to make it works wirelessly! The problem comes from the esp itself ? And not from snapcast ?
If I can help in any way, it will be a pleasure!

jorgenkraghjakobsen, 2020-04-23:
Yes the UDP wireless packet loss is not god - at least not here in my setup. 
Help is much appreciated. I need to back port some code to this repo and / or turn some of my common used component into sub component. Long list of todo s. 
What is your setup - what i2s amplifier are you using we may need to abstract that if i can not get you on my ma12070p amps. 
/j

dmckminard, 2020-04-23:
Yes, and I see that you're active!
I'm running with PCM5102

jorgenkraghjakobsen, 2020-04-23:
Hi minard 
I just dusted of my setup and I get  steaming audio through - not as bad as I am use to. I do get some distortion that i think ordinate from direct with to i2s driver. Will add a ring buffer. :-)

dmckminard, 2020-04-23:
Nice ! Can you upload it so I can test ?

jorgenkraghjakobsen, 2020-04-24:
Ring buffer added - Added my dsp_processor component, another side project that added generic high level audio processing to the party. 
The dsp_processor is also very much in work in progress with a long feature list. Master plan will be a dynamic environment where the full audio flow can be configured and tuned from an web interface. 
If you want to kick in then have a look at the time stamping that is need to keep client in sync. 

Uploaded in 5 min /J

dmckminard, 2020-04-24:
A web interface would be very cool ! 
Which sound card are you using ? also trying in WLAN ?

jorgenkraghjakobsen, 2020-04-24:
Yes - Did some nice progress last night. Quite stable client using WLAN - the client here is wlan only until now. 
I do chip design in my day time job - so i am using my own I2S input class D audio amplifier. MA12070P - We have a nice pi hat that will be a good fit for you Raspberry pi project as well. 
/j

jorgenkraghjakobsen, 2020-04-24:
Fixed major mistake - Works perfect on WLAN - rock stable. 
I have not implemented any buffer / throttle control - for multi room streaming that is not good. But for single point audio streaming snapcast is the best i have seen until now. 
/J

dmckminard, 2020-04-24:
Ok very very nice, does the buffer part isn’t managed by snapcast ? We need to implement this in this program ?

jorgenkraghjakobsen, 2020-04-24:
No the snap server will just time stamp outgoing audio packages and all client are responsible to play samples at  t+500ms  - where the 500ms is communicated to all clients from server. 
Volume and mute function are passed around instantly and this gives a impression for a very responsive system.

NathanTct, 2020-04-28:
Hi there ! I am actually trying to run your code, because it sounds pretty cool, but for now I can't hear any sound ! I guess it would be  a audio format problem, what do you recommand to use ? I am on an ESP32S coupled with a PCM5102 ! I succeeded to flash your code on the ESP but can't read any sound. If you have an idea or if you know the issue, a bit of help would be really appreciated ! And i'd like to help you on this project, I think this could be great for multirooming

jorgenkraghjakobsen, 2020-04-28:
Hi Nathan 
From what i see on the data sheet of PCM5102 - you just need to keep SCK grounded and route in BCK, WS, and DATA from the ESP32 I2S interface. 
I use the MA12070P from Infineon  - Just need to hookup I2C and I2S and it flys /J

jorgenkraghjakobsen, 2020-04-28:
For PCM5102 Use the setup from https://hackaday.io/project/166122/instructions by Raphael H. 
You need to set the correct i2s pins in the component/dsp_processor/dsp_processor.c line 60 pin_config0 
I will add a menuconfig for that component - there are other cool stuff there :-)

jorgenkraghjakobsen, 2020-04-28:
I have added i2s config to project menu

NathanTct, 2020-04-29:
Perfect, I saw it, it would help me a lot ! Thank you !

TekniskSupport, 2020-04-29:
> Perfect, I saw it, it would help me a lot ! Thank you !
@NathanTct Did you get it working using the PCM5102? 
I'm looking at using the same hardware.

TekniskSupport, 2020-06-28:
@NathanTct did you get it up and running?

I've fiddled around a bit with the PCM5102 and an ESP32 the last few days, and seems to get it up and running, but no sound output.

```
Settime from sntp
I (6708) SNAPCAST: Initializing SNTP
I (6708) SNAPCAST: Waiting for system time to be set... (1/10)
I (8708) SNAPCAST: The current date/time in UTC is: Sun Jun 28 16:56:49 2020
Called
I (8708) I2S: queue free spaces: 7
I (8708) I2S: DMA Malloc info, datalen=blocksize=3840, dma_buf_count=8
I (8708) I2S: PLL_D2: Req RATE: 48000, real rate: 48076.000, BITS: 32, CLKM: 13, BCK: 4, MCLK: 12292917.167, SCLK: 3076864.000000, diva: 64, divb: 1
Ringbuffer ok
I (8728) SNAPCAST: Connected to AP
I (8728) SNAPCAST: ... allocated socket
Called http
I (9038) SNAPCAST: ... connected
I (9058) SNAPCAST: Setting volume: 100
I (9068) SNAPCAST: Received codec header message

I (9068) SNAPCAST: Codec : opus , Size: 12 

I (9068) SNAPCAST: Opus sampleformat: 48000:16:2

I (9078) SNAPCAST: Initialized opus Decoder: 0
I (9078) SNAPCAST: OPUS encoding buffer too small, resizing to 120 samples per channel
I (9088) SNAPCAST: OPUS encoding buffer too small, resizing to 240 samples per channel
I (9098) SNAPCAST: OPUS encoding buffer too small, resizing to 480 samples per channel
I (9108) SNAPCAST: OPUS encoding buffer too small, resizing to 960 samples per channel
I (9138) SNAPCAST: BaseTX     : 1821325 978012 
I (9138) SNAPCAST: BaseRX     : 1593363410 140694 
I (9138) SNAPCAST: baseTX->RX : 1591542 s 
I (9138) SNAPCAST: baseTX->RX : -837ms 
I (9148) SNAPCAST: Latency : -1591542085.907 
I (10098) SNAPCAST: BaseTX     : 1821326 978719 
I (10098) SNAPCAST: BaseRX     : 1593363411 106767 
I (10098) SNAPCAST: baseTX->RX : 1591542 s 
I (10108) SNAPCAST: baseTX->RX : -871ms 
I (10108) SNAPCAST: Latency : -1591542085.891 
```