jorgenkraghjakobsen/snapclient #2: Code can not be compiled

karimSad, 2020-05-22:
Hallo,

I am working now with ESP32 and I am trying to compile your code but unfortunately it is not working.
I tried ESP-IDF 3.x and also ESP-IDF 4.0 but the same error is staying

`/main/main.c:25:10: fatal error: opus.h: No such file or directory`

Though the component "libopus" has this header file. It would be nice if you can give me a tip if you know where the problem might be.

Regards,
Karim

jorgenkraghjakobsen, 2020-05-22:
Hi Karim - I only tested on 3.x with old make system. 
I have experienced the same issue related to build system not sourcing components/include correct on some of my old repos as well. I will look in to this. /j

karimSad, 2020-05-22:
Hi Jorgen,

thanks for your fast answer. 
Now the code is compiled without any problem. The idea was that I was trying to run your code in the "ninja" build system which is "default" I would say in esp-idf framework "idf.py build".

It would be nice in the future to configure the project with CMakeLists.txt and can be compiled in ninja build system or by the command "idf.py build"

Regards,
Karim

jorgenkraghjakobsen, 2020-05-22:
Hi Karin - I did my first ninja build this week - Yes I need to update all my repos to be compatible with new build system. I am very positive on a more structured way of dealing with esp32 components but my focus has not been there yet. 
Let me know if you get it up running. /J

karimSad, 2020-05-22:
Thanks again for your fast response. Well, if you mean you get it running with ninja this week or you started with? If the case is yes, it would be very helpful for me if you push your changes to the repo to be able to run the project under ninja.

Now it is working on my machine with make but not yet with ninja.

Regards,

karimSad, 2020-05-26:
Hi Jorgen,

Thank for the support of CMake. 
I am relativ new to Github, so I don't know if this issue should be closed from your side or my side. 

PS: You made CMake support for 4.x and this very nice. But also, snapclient related applications may work under 3.x because of ESP-ADF Audio support framework from espressif. Anyway I still have some problems in my machine. When I am free I would pullrequest the CMake support for ESP-IDF 3.x.

Thank for your work in this project.

Regards,

jorgenkraghjakobsen, 2020-05-26:
Hi Karim
You need to update you check out using 
git pull 
git submodule update --init 
last command will pull in the components/opus from the xiph/opus repo 
compile using idf.py build 
Build system will complain on a non init variable: 
Fix it by adding = 0 to line 54 in snapclient/components/opus/opus/silk/quant_LTP_gains.c
opus_int32           res_nrg_Q15_subfr, res_nrg_Q15 = 0, rate_dist_Q7_subfr, rate_dist_Q7, min_rate_dist_Q7;
My snapcast clinet is not using any ESP-ADF stuf - Pure ESP-IDF :-)
/j

karimSad, 2020-06-09:
Well, from my side this issue is solved since a while. So, I will close it. Thanks again

closed 2020-06-09 by karimSad