jorgenkraghjakobsen/snapclient #4: Edit the code to work on different Hardware Platform

karimSad, 2020-06-07:
Hi Jorgen,

I have some questions that would help me to run your nice repo on my standard development board [ESP32-LyraT V4.3).](https://docs.espressif.com/projects/esp-adf/en/latest/design-guide/board-esp32-lyrat-v4.3.html)

In Lyra board, an internal codec [ES8388](http://www.everest-semi.com/pdf/ES8388%20DS.pdf) and internal power amplifier (instead of MerusAudio) are used in the Lyra board.

The internal codec of the Lyra board is connected to the esp32-wrover module via i2s interface for audio and i2c for control.

My approach to let my board work with snapclient was:

- I commented code related to MerusAudio

- I initialized the codec from ready-to-use APIs from ESP-ADF (PS: ESP-ADF now supports ESP-IDF 4.0 and 4.1  :) )

- I choosed the i2s pins which should work on my board

The code was compiled without problems. But when I flash the code, I am getting weird behavior.

My questions: (also understanding questions) :))

- Could you please give me tips, how the workflow might be to let the snapclient work on another hardware platform which has for example an internal codec and power amplifier such as Lyra.  

- The functionality of the codec in your repo is done in opus submodule, that means you are not using a codec-chip?

Thanks in advanced for your answer

Regards,

jorgenkraghjakobsen, 2020-06-07:
Hi Karim 
Sure - I have 2 older Lyra boards here but never put much attention - time you known and the fact that I have first hand access to codec/audio amplifiers.
I my setup i have a codec build into my audio power amplifier. Here codec is only and digital to analog converter that is integrated into the audio power amplifier. 
On the Lyra board the codec is a real codec that can do analog to digital for mic and aux input and digital to analog for the two analog input audio amplifiers. It is a bit old style as microphones and amplifers do digital I2S interface directly to the esp32. 
On the Lyra setup your approach is correct - call the appropriate Lyra ES8388 setup instead of my setup_ma12070 and you should be good to go. 
One thing that needs to be aligned is the I2S data format from ESP32 to the ES8388 - this is handled in dsp_processor.c 
For MA12070P we are using 32bits/48000kHz - But I recall my early Lyra experiments ran default 16bits/48000kHz 
Hope I pointed you in some direction. /j

karimSad, 2020-06-09:
Hi Jorgen,

thanks a lot for your quick respone. Another question is :)

Now I can connect to snapserver and I am getting the following:

`
Setup ringbuffer using PSRAM
Buffer_struct ok
Buffer_stoarge ok
Ringbuf ok
Ringbuffer ok
I (10256) SNAPCAST: Connected to AP
I (10266) SNAPCAST: Enable mdns
I (10286) SNAPCAST: Lookup snapcast service on netowork
I (10376) SNAPCAST: SNTP time set from server number :1
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
I (14376) SNAPCAST: Found 4902a8c0
I (14376) SNAPCAST: ... allocated socket
I (14386) SNAPCAST: ... connected
I (14396) SNAPCAST: Buffer length:  1000
I (14396) SNAPCAST: Ringbuffer size:192000
I (14396) SNAPCAST: Latency:        0
I (14396) SNAPCAST: Mute:           0
I (14406) SNAPCAST: Setting volume: 100
I (14416) SNAPCAST: Received codec header message

I (14416) SNAPCAST: Codec : flac , Size: 1362

I (14416) SNAPCAST: Opus sampleformat: 570425344:32772:32772

I (14426) SNAPCAST: Failed to init opus coder
I (14436) SNAPCAST: Initialized opus Decoder: -1
E (14436) SNAPCAST: Decode error : -1

wait ... buffer : 0
E (14466) SNAPCAST: Decode error : -1

E (14486) SNAPCAST: Decode error : -1

E (14496) SNAPCAST: Decode error : -1

E (14516) SNAPCAST: Decode error : -1

E (14526) SNAPCAST: Decode error : -1

E (14556) SNAPCAST: Decode error : -1

E (14576) SNAPCAST: Decode error : -1

E (14616) SNAPCAST: Decode error : -1

E (14636) SNAPCAST: Decode error : -1

`

So, from menuconfig -> ESP32 audio buffer and I2s pin config 
I did the following configs:
Master i2s bck = 5
Master i2s data lrck = 25
Master i2s data out = 26
Slave i2s bck = 5
Slave i2s data lrck = 25
Slave i2s data out = 35

I think these configs are not used 

from components/rtprx/rtprx.c
I changed i2s pin configs as follows:
`i2s_pin_config_t pin_config = {
.bck_io_num   = 5, //CONFIG_EXAMPLE_I2S_BCK_PIN,
.ws_io_num    = 25, //CONFIG_EXAMPLE_I2S_LRCK_PIN,
.data_out_num = 26, //CONFIG_EXAMPLE_I2S_DATA_PIN,
.data_in_num = 35                                                       //Not used
};`
I commented the following code lines:
`ma_write(0x20,1,0x0040,cmd,1);` and `ma120_setup_audio(0x20);`

**Questions :)**
1- Do you think this is the right pin configs or am I missing something. I am some how here not understanding well, why in your case you are using two cases (master and salve) but in Lyra they are using just four pins for i2s interface

2- In Lyra Board there is, as said before, an internal codec. But in this repo there are built-in decoder  opus_decoder_create. So in my case, i must output the audio direct from the DAC and i should avoid pass ringbuffer through the codec which will decode the decoded data again. Do you think am i right or i am missing something.

3- To use Ringbuffer from main.c: 
I used this line in main.c `RingbufHandle_t s_ringbuf_i2s = NULL;` and in components/dsp_processor/dsp_processor.c I replaced this line `static RingbufHandle_t s_ringbuf_i2s = NULL;` by this line `extern RingbufHandle_t s_ringbuf_i2s;`
Then I called s_ringbuf_i2s  from ADF APIs from main.c.
Do you think is approach right? 

Regards,

karimSad, 2020-06-11:
Hi Jorgen,

very happy for these news :)
I am not near to my board. Tomorrow I would test it on my Lyra board. Thanks a lot.

Best Regards,
Karim

karimSad, 2020-06-13:
Hi Jorgen,

well, unfortunatelly the snapclient on Lyra is not working yet. There is no error but it can not connect to the snapserver i created on my machine. I tested my snapserver with snapdroid app from google store and it is working.
from the console in my snapserver i am not getting the "hello" message from the Lyra (client). 

My terminal from Lyra after connecting to the wifi AP is as follows:
`Setup ringbuffer using PSRAM
Buffer_struct ok
Buffer_stoarge ok
Ringbuf ok
Ringbuffer ok
I (8708) SNAPCAST: Connected to AP
I (8708) SNAPCAST: Enable mdns
I (8718) SNAPCAST: Lookup snapcast service on netowork
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
I (11068) SNAPCAST: SNTP time set from server number :1
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
W (11818) SNAPCAST: No results found!
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
I (12818) SNAPCAST: Lookup snapcast service on netowork
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
W (15918) SNAPCAST: No results found!
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
I (16918) SNAPCAST: Lookup snapcast service on netowork
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
W (20018) SNAPCAST: No results found!
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
I (21018) SNAPCAST: Lookup snapcast service on netowork
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
W (24118) SNAPCAST: No results found!
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
I (25118) SNAPCAST: Lookup snapcast service on netowork
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
I (26128) SNAPCAST: SNTP diff us: 277216 , 292276
I (26128) SNAPCAST: SNTP diff us: -15.06
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
`

I gave the host IP of my snapserver and port = 1704. I tried also port = 1705 and port =1780 (http stream)

So, it is silly question:)) you were able to connect to snapserver?

Thanks in advance
Karim

jorgenkraghjakobsen, 2020-06-13:
Hi This is the log of my snapserver 
./snapserver 
`2020-06-13 22-35-48.483 [Info] (main) Adding stream: pipe:///tmp/snapfifo?name=default
2020-06-13 22-35-48.484 [Notice] (init) Settings file: "/home/jakobsen/.config/snapserver/server.json"
2020-06-13 22-35-48.501 [Info] (Avahi) Adding service 'Snapcast'
2020-06-13 22-35-48.516 [Info] (start) Creating TCP acceptor for address: 0.0.0.0, port: 1705
2020-06-13 22-35-48.517 [Info] (start) Creating HTTP acceptor for address: 0.0.0.0, port: 1780
2020-06-13 22-35-48.517 [Info] (PcmStream) PcmStream sampleFormat: 48000:16:2
2020-06-13 22-35-48.518 [Info] (PcmStream) metadata={
2020-06-13 22-35-48.518 [Info] (PcmStream)     "STREAM": "default"
2020-06-13 22-35-48.518 [Info] (PcmStream) }
2020-06-13 22-35-48.518 [Info] (onMetaChanged) onMetaChanged (default)
2020-06-13 22-35-48.518 [Info] (PipeStream) PipeStream mode: create
2020-06-13 22-35-48.518 [Info] (start) Stream: {"fragment":"","host":"","path":"/tmp/snapfifo","query":{"chunk_ms":"20","codec":"opus","name":"default","sampleformat":"48000:16:2"},"raw":"pipe:////tmp/snapfifo?chunk_ms=20&codec=opus&name=default&sampleformat=48000:16:2","scheme":"pipe"}
2020-06-13 22-35-48.519 [Info] (initEncoder) Opus bitrate: 192000 bps, complexity: 10
2020-06-13 22-35-48.522 [Error] (PosixStream) Exception: end of file
2020-06-13 22-35-48.522 [Info] (start) Creating stream acceptor for address: 0.0.0.0, port: 1704
2020-06-13 22-35-48.522 [Info] (main) number of threads: 4, hw threads: 4
2020-06-13 22-35-49.568 [Info] (Avahi) Service 'Snapcast' successfully established.
2020-06-13 22-35-52.443 [Notice] (handleAccept) StreamServer::NewConnection: 192.168.1.233
2020-06-13 22-35-52.453 [Info] (onMessageReceived) Hello from 30:AE:A4:45:EE:FC, host: ESP32-Caster, v0.0.2, ClientName: libsnapcast, OS: esp32, Arch: xtensa, Protocol version: 2
2020-06-13 22-37-57.008 [Info] (onStateChanged) onStateChanged (default): 2
2020-06-13 22-38-04.291 [Info] (onStateChanged) onStateChanged (default): 1
2020-06-13 22-38-18.469 [Notice] (handleAccept) StreamServer::NewConnection: 192.168.1.233
2020-06-13 22-38-18.477 [Info] (onMessageReceived) Hello from 30:AE:A4:45:EE:FC, host: ESP32-Caster, v0.0.2, ClientName: libsnapcast, OS: esp32, Arch: xtensa, Protocol version: 2
`
Last 4 lines is my esp32 client connction 

Your setup is looping in this code 
` while ( !r || err )
        {  ESP_LOGI(TAG, "Lookup snapcast service on netowork");
           esp_err_t err = mdns_query_ptr("_snapcast", "_tcp", 3000, 20,  &r);
           if(err){
             ESP_LOGE(TAG, "Query Failed");
           }
           if(!r){
             ESP_LOGW(TAG, "No results found!");
           }
           vTaskDelay(1000/portTICK_PERIOD_MS); 
        } 
     `
Try to comment that out and hardcode ip of your snap server here 
` servaddr.sin_family = AF_INET;
        servaddr.sin_addr.s_addr = r->addr->addr.u_addr.ip4.addr;         // inet_addr("192.168.1.158");
        servaddr.sin_port = htons(r->port);
`
/j

jorgenkraghjakobsen, 2020-06-13:
I have checked in my snapcast server conf in the directory snapcast 
/j

karimSad, 2020-06-13:
thanks a lot. I would give my feedback here as I get my board runnging.

jorgenkraghjakobsen, 2020-06-13:
Also snap volume control to the ADF hal abstration - now need to do a hal driver for my own amp :-)

karimSad, 2020-06-14:
well, at least i think that i know where is the problem :=)) 
I am using Ubuntu under windows as WSL. And the Mdns service seems not to work yet correctly. [Here  is](https://github.com/microsoft/WSL/issues/384) the link

karimSad, 2020-06-16:
Hi Jorgen,

due to several network problems was my board not working. Now it is working but for maximum 30 seconds then it makes a new restart :)

karimSad, 2020-06-16:
Hallo Jorgen,

a short question :)
In your Lyra board or even in your custom board, the audio is working fine for a relative long time or the program is falling down and RTOS is restarting again?

In my case, it is almost not working. So i wanted to know if it is from my wifi network.

jorgenkraghjakobsen, 2020-06-16:
My code is rocking - Is your snapserver on a linux instance running under windows. Can you try to test it vs a client on a nother pc. 
My client keep on for ever. Do you get 30 sec of audio or no audio at all? 
/j

karimSad, 2020-06-17:
The 30 sec is kind of rare to get. My snapclient on the android is woking fine. So, I think that the wifi configs by me on the esp32 target are some how not set correctly. 

My snapserver is working now on Raspbian OS in Raspberry pi.

karimSad, 2020-06-17:
this is the terminal output after wifi init
I also changed pcm_size from 120 to 960 in main.c

`I (7640) esp_netif_handlers: sta ip: 192.168.2.104, mask: 255.255.255.0, gw: 192                                                                                                                                                             .168.2.1
I (7640) SNAPCAST: got ip:192.168.2.104
I (7640) SNAPCAST: connected to ap SSID:kontor password:1234
I (7650) system_api: Base MAC address is not set, read default base MAC address                                                                                                                                                              from BLK0 of EFUSE
Settime from sntp
I (12660) SNAPCAST: Initializing SNTP
Called
I (12660) I2S: queue free spaces: 7
I (12660) I2S: DMA Malloc info, datalen=blocksize=1920, dma_buf_count=8
I (12680) I2S: APLL: Req RATE: 48000, real rate: 47999.961, BITS: 16, CLKM: 1, B                                                                                                                                                             CK_M: 8, MCLK: 12287990.000, SCLK: 1535998.750000, diva: 1, divb: 0
Setup ringbuffer using PSRAM
Buffer_struct ok
Buffer_stoarge ok
Ringbuf ok
Ringbuffer ok
I (12690) SNAPCAST: Connected to AP
I (12700) SNAPCAST: Enable mdns
I (12700) SNAPCAST: Lookup snapcast service on netowork
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
I (15040) SNAPCAST: SNTP time set from server number :1
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
I (16800) SNAPCAST: Found 1e02a8c0
I (16800) SNAPCAST: ... allocated socket
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
I (17370) SNAPCAST: ... connected
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
wait ... buffer : 0
I (20680) SNAPCAST: Buffer length:  1000
I (20680) SNAPCAST: Ringbuffer size:192000
I (20680) SNAPCAST: Latency:        0
I (20680) SNAPCAST: Mute:           0
I (20680) SNAPCAST: Setting volume: 100
I (20690) SNAPCAST: Received codec header message

I (20690) SNAPCAST: Codec : opus , Size: 12

I (20700) SNAPCAST: Opus sampleformat: 48000:16:2

I (20700) SNAPCAST: Initialized opus Decoder: 0
I (20710) SNAPCAST: OPUS encoding buffer too small, resizing to 960 samples per                                                                                                                                                              channel
I (22360) I2S: Chunk :960 105600 118380
I (22370) I2S: Chunk :960 104640 117780
I (22380) I2S: Chunk :960 103680 117180
I (22390) I2S: Chunk :960 106560 116640
I (22400) I2S: Chunk :960 105600 116100
I (22410) I2S: Chunk :960 108480 115620
I (22420) I2S: Chunk :960 111360 115200
I (22430) I2S: Chunk :960 110400 114780
I (22440) I2S: Chunk :960 109440 114360
I (22450) I2S: Chunk :960 108480 113940
I (22460) I2S: Chunk :960 107520 113520
I (22470) I2S: Chunk :960 106560 113100
I (22480) I2S: Chunk :960 105600 112680
I (22490) I2S: Chunk :960 104640 112260
I (22500) I2S: Chunk :960 103680 111840
I (22510) I2S: Chunk :960 102720 111420
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandle                                                                                                                                                             d.
Core 0 register dump:
PC      : 0x400d9617  PS      : 0x00060930  A0      : 0x8008e7a4  A1      : 0x3f                                                                                                                                                             fe7800
A2      : 0x00000000  A3      : 0x00000158  A4      : 0x000000f0  A5      : 0x00                                                                                                                                                             000000
A6      : 0x00000000  A7      : 0x000057ee  A8      : 0x00000000  A9      : 0x3f                                                                                                                                                             fe77e0
A10     : 0x000008cb  A11     : 0x3ffb0cbc  A12     : 0x3f80013c  A13     : 0x3f                                                                                                                                                             ff0820
A14     : 0x00000000  A15     : 0x3f800130  SAR     : 0x0000001f  EXCCAUSE: 0x00                                                                                                                                                             00001c
EXCVADDR: 0x00000000  LBEG    : 0x4008a6b8  LEND    : 0x4008a6e6  LCOUNT  : 0x00                                                                                                                                                             000000

ELF file SHA256: e4a826abdfe41079

Backtrace: 0x400d9614:0x3ffe7800 0x4008e7a1:0x3fff0a60
`

Thenicolaibulow, 2020-06-17:
Hi guys, 

I just wanted to chime in, and point out, that i'm having a very similar problem. 
At times my setup (Server running on ubuntu with pulse | /tmp/snapfifo, and two custom boards as my clients) does something quite similar. 

Some times it'll run nicely for minuts, and then suddenly, without further ado, just crash. 
After playing quite a bit around with it, i've noticed some important factors: 

- Check the server config - I'm running buffer = 150ms, and chunk size = 20ms. 
- I changed my servers wifi uplink to 5GHz, which helped significantly. 

At first my setup (just like Karimsad's) would crash, after the opus decodes was increasing in size. - It doesn't as often anymore, and this was on a earlier build anyways. I figure it's stille a buffer issue? It seems, that is the decoder still isn't happy, after the pcm_size is > 960, it'll try and double the size again, crashing the system. 

Now i'm running into the same problem af karimsad. Whenever i turn off one client the other one dies every time, and panic's and dsp_processor.c line ~170. 
This is where Jørgen is doing the 16 - > 32 bit conversion, which i need. 

I hope this might help you out a bit.

karimSad, 2020-06-17:
Hi Nicolai,

but how did you changed to uplink of 5 GHz though the ESP32 supports (as I know) only 2.5 GHz. Or you are using another Hardware Platform?

Thenicolaibulow, 2020-06-17:
Yes, you're right (unfortunately..) the esp only supports 2.4GHz, however the machine running my server is capable of doing 5GHz. It was very apparent, that drastically reduced the amount of lost packages when I change the server wifi. 

As I've just moved into a new apartment, I wasn't aware of just how bad the wifi interference is here. The 2.4GHz band seriously crowded. Luckily, very few of my neighbors are using 5GHz wifi! :D

jorgenkraghjakobsen, 2020-06-17:
Hi N and K 
The minimum buffer value that can be set at the snap server is 400 ms - Set it lower it will minimize to 400 ms. On the client side I have implemented a switch to run with or with out PSRAM. On non PSRAM esp32 module the server buffer setting is ignore and hardcode to 150ms. 
And yes it seems that you are experience WLAN network errors - I also had that issues when my server was on wifi as well. 
My problems dissolved after I hooked my workstation/server on my wired network. I have optimized my WLAN access point and my esp32 config for low latency - there is a concept called pkg aggregation that bundle smaller packet in bigger pkg to optimize for through put - but we want low latency. 
My WLAN access point is ZyXEL NWA3560-n 
Here I have disabled A-MPDU Aggregation and A-MSDU Aggregation
Like wise on the ESP32 WiFi I have disable 
[ ] WiFi AMPDU TX
[ ] WiFi AMPDU RX
under components/wifi/ in menuconfig
This help out a lot on my RTP low latency UDP audio streaming I was working with prior to the Snap stuff. 

You mite want to do this optimizations as well.  

Odd problem with one client taken the other one down as well. 
As understand each client has its own communication and audio flow with the server. Removing one would give more bandwidth to the other. I have only ran max 3 client, one on my workstation, one on my android and the esp32 client. 
/j

Thenicolaibulow, 2020-06-17:
Aha! 

Thanks for clarifying, Jørgen. 
I'll take a look at my esp-wifi & router settings! We can agree that it sounds odd that one client would take down the other - what connection to one another would they have?? I'll post a client and server log. - maybe you'll notice something, that I've overlooked.

othiman, 2021-09-30:
Hi.

I try to compile and run snapclient for the ESP32_LyraT 4.3 board.

What I have done so far (under MacOS Big Sur):
- Get recent version of idf and install it (master branch)
- Get recent version of adf and set ADF_PATH (master branch)
- Clone the snapclient Github repo and run `idf.py menuconfig`
- Change WLAN credentials and Snapcast server IP etc
- Change Audio HAL to ESP32-Lyrat V4.3 board
- Exit and save menuconfig
- Run `idf.py build` (and fix an compiler error in opus lib, seems to MacOS specific)
- Run `idf.py flash -p /dev/cu.SLAB_USBtoUART monitor`
- Waiting for flashing done and manually reset the device via RST button

After that I get the log from the end of this post via idf.py monitor. That looks quite OK, but I do not hear any sound from the connected headphones. Two things look suspicious to me:
- The i2s_check_set_mclk seems to try to set an uninitailsed number on startup and therefore an error is logged.
- While playing data from snapcast server I get `Error readding audio from ring buf` messages.

Can someone give me a hint what I am doing wrong?

Best regards,
Thomas

```
--- idf_monitor on /dev/cu.SLAB_USBtoUART 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6724
load:0x40078000,len:14908
load:0x40080400,len:3792
0x40080400: _init at ??:?

entry 0x40080694
I (27) boot: ESP-IDF v4.4-dev-3235-g3e370c4296 2nd stage bootloader
I (27) boot: compile time 23:36:59
I (27) boot: chip revision: 1
I (31) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (38) boot.esp32: SPI Speed      : 40MHz
I (43) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 4MB
I (52) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length
I (68) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (76) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (83) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (91) boot:  3 factory          factory app      00 00 00010000 00100000
I (98) boot:  4 ota_0            OTA app          00 10 00110000 00100000
I (106) boot:  5 ota_1            OTA app          00 11 00210000 00100000
I (113) boot: End of partition table
I (118) boot: Defaulting to factory image
I (122) boot_comm: chip revision: 1, min. application chip revision: 0
I (130) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=235d8h (144856) map
I (191) esp_image: segment 1: paddr=00033600 vaddr=3ffb0000 size=03940h ( 14656) load
I (197) esp_image: segment 2: paddr=00036f48 vaddr=40080000 size=090d0h ( 37072) load
I (212) esp_image: segment 3: paddr=00040020 vaddr=400d0020 size=a7f38h (687928) map
I (461) esp_image: segment 4: paddr=000e7f60 vaddr=400890d0 size=120c0h ( 73920) load
I (492) esp_image: segment 5: paddr=000fa028 vaddr=50000000 size=00010h (    16) load
I (505) boot: Loaded app from partition at offset 0x10000
I (505) boot: Disabling RNG early entropy source...
I (517) psram: This chip is ESP32-D0WD
I (519) spiram: Found 64MBit SPI RAM device
I (519) spiram: SPI RAM mode: flash 40m sram 40m
I (522) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
[... 154 lines cut ...]
I (32491) I2S: Chunk :1920 60 ms
I (33591) I2S: Chunk :1920 35 ms
I (34661) I2S: Chunk :1920 40 ms
I (35731) I2S: Chunk :3840 54 ms
I (35831) I2S: Error readding audio from ring buf : read 732 of 1920 , missing 1188
I (35831) I2S: Read the next 1188 
I (36811) I2S: Chunk :1920 46 ms
I (37901) I2S: Chunk :1920 44 ms
I (38971) I2S: Chunk :1920 27 ms
I (39821) I2S: Error readding audio from ring buf : read 528 of 1920 , missing 1392
I (39821) I2S: Read the next 1392 
I (40051) I2S: Chunk :1920 69 ms
I (41141) I2S: Chunk :1920 48 ms
I (42181) I2S: Chunk :1920 16 ms
I (43211) I2S: Chunk :1920 21 ms
I (43681) I2S: Error readding audio from ring buf : read 336 of 1920 , missing 1584
I (43691) I2S: Read the next 1584 
I (44241) I2S: Chunk :1920 2 ms
I (45311) I2S: Chunk :1920 48 ms
I (46391) I2S: Chunk :1920 53 ms
I (47471) I2S: Chunk :1920 14 ms
I (47631) I2S: Error readding audio from ring buf : read 132 of 1920 , missing 1788
I (47631) I2S: Read the next 1788 
I (48551) I2S: Chunk :1920 43 ms
```