jorgenkraghjakobsen/snapclient #5: Bad Base Message parsing?

AsbjornOlling, 2020-06-15:
Hi - I think you might be parsing base messages wrong.

If you look at the `base_message_serialize` and `base_message_deserialize`  functions, I think you mixed up the order of the sent time and received time fields.

https://github.com/jorgenkraghjakobsen/snapclient/blob/c93dacc4836979b5823914e77a7eb7bc1e788feb/components/lightsnapcast/snapcast.c#L19

If you look at the documentation for the Base message data, they say that they are reverse order.

https://github.com/badaix/snapcast/blob/master/doc/binary_protocol.md

I haven't written a fix, or tested the code, so I don't really know for sure. Just wanted to give you a head up :)

jorgenkraghjakobsen, 2020-06-15:
Hi Asbjørn - Thank is very likely. The protocol message parsing was done by Bridadan how did a prove of concept. 
And yes I my time sync with server as all ways troubled me - just look at the messy code. I will  look it to this on my next late night session. - Tonight is gone :-) I was playing around with real time tuning on the APLL and found that that is the key to next level of multi channel sync. 
Very happy to get you help on the project - Thanks - I you have more time to spent let me know. 
/j

jorgenkraghjakobsen, 2020-06-16:
Hi Asbjørn 
No - document was wrong - Just looked up the code from badaix/snapcast/common 
    https://github.com/badaix/snapcast/blob/master/common/message/message.hpp 
The ser/deser done as describer here. 

I guess the time keeping library function on each platform differ. Have not looked more in to this - but have to sometime into the future. 
But thanks - Will you late Badaix know about the documentation 
/Jørgen