jorgenkraghjakobsen/snapclient #7: Compile error on current master

TekniskSupport, 2020-06-26:
Finally got my hands on some hardware!
On the current master (c93dacc4836979b5823914e77a7eb7bc1e788feb) i get compile error, it worked on
2b9dd956cf4b980046c427efb29b6f4227cf3a21 

I get 

```
App "snapclient" version: c93dacc-dirty
CC build/main/main.o
/home/iesus/esp32snapclient/main/main.c:19:10: fatal error: board.h: No such file or directory
 #include "board.h"
          ^~~~~~~~~
compilation terminated.
/home/iesus/esp/esp-idf/make/component_wrapper.mk:291: recipe for target 'main.o' failed
make[1]: *** [main.o] Error 1
/home/iesus/esp/esp-idf/make/project.mk:635: recipe for target 'component-main-build' failed
make: *** [component-main-build] Error 2
```

Tried updating esp-idf but no luck so far.

You've refactored quite a lot since so i'm having trouble finding the cause (`git diff 2b9dd956cf4b980046c427efb29b6f4227cf3a21|wc -l`is > 67000)

constant-flow, 2020-09-11:
I also had issues to get [c93dacc](https://github.com/jorgenkraghjakobsen/snapclient/commit/c93dacc4836979b5823914e77a7eb7bc1e788feb) running. I finally got [b063b3](https://github.com/jorgenkraghjakobsen/snapclient/commit/b063b3d7fd0442e1d1e084ee2f21ce249bec8645) to compile and run on my wrover and wroom boards.

For me the trick was to install `esp-adf` next to `esp-idf`. The latest changes seem to target a specific set of hardware (e.g. lyrat ADF board).

I think having branches for every set of HW would be cool, so the master does i2s-out for the basic wrover/wroom boards to configurable pins and other branches are more advanced and optimized towards their HW/connections.

Doing so would make it easy for all the starters to get on hands on this great project :)

jorgenkraghjakobsen, 2020-09-29:
Hi - Sorry yes i have had time for this project for a long time. 
I also fund the code broken and had to rool back to IDF 4.1 and ADF 1.1 - Code work with my up date from yesterday. 
/j

jorgenkraghjakobsen, 2020-09-29:
What hardware are you now - last time we spoke you was on lyra 3.4? I got my lyra 3.4 up running with only volume contorl on one channel - will look at that now. 
And I will try to do a menuconfig that cover all hardware setup :-)

grewhit25, 2020-11-23:
Just to say thank you for this great project.
Also to note that @constant-flow workaround above worked for me - lyrat 4.3

AxelJust, 2020-12-20:
Hey,
I would recomend to look in to this Project: https://github.com/sle118/squeezelite-esp32 
It seems that they get throug the Calanges quite well and already suport multiple features. It might be easyer to implement a snapclient in there.
Greating

jorgenkraghjakobsen, 2020-12-20:
Hi Alex - 
Yes - squeeselite - esp32 is a very cool project - hopefully we can squeeze in a snapclient ones we master it. :-) 
/J

TekniskSupport, 2021-04-05:
This issue is way old, closing

closed 2021-04-05 by TekniskSupport