jorgenkraghjakobsen/snapclient #11: Compilation on windows?

tocklime, 2020-10-10:
Is windows compilation supposed to work?

I've cloned the repo, done the git submodules, installed StrawberryPerl and esp-idf-tools with chocolatey.

When I run idf.py build from the checkout in the ESP-IDF terminal I get some output that ends with:

```shell
-- Configuring incomplete, errors occurred!
See also "C:/Users/greg/Documents/GitHub/snapclient/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1
```

I've attached the CMakeOutput.log, but I can't see any obvious things going wrong.

[CMakeOutput.log](https://github.com/jorgenkraghjakobsen/snapclient/files/5358723/CMakeOutput.log)

tocklime, 2020-10-10:
...nope, not windows specific. I get a very similar error in windows subsystem for linux running ubuntu 20.04 and native manjaro.

tocklime, 2020-10-11:
ah, i got past configuration (and into build errors) by setting ADF_PATH to my checkout of esp-adf. If that's the right thing to do then maybe document it?

pmumby, 2020-10-12:
What specific build errors are you seeing? I have an issue in my checked out code with the opus submodule... Specifically it fails to compile due to a compiler error about an un-initialized variable.

The following diff of the file `components/opus/opus/silk/quant_LTP_gains.c resolves the issue (by initializing the res_nrg_Q15 variable):
```
--- a/silk/quant_LTP_gains.c
+++ b/silk/quant_LTP_gains.c
@@ -61,6 +61,8 @@ void silk_quant_LTP_gains(
     /***************************************************/
     min_rate_dist_Q7 = silk_int32_MAX;
     best_sum_log_gain_Q7 = 0;
+    res_nrg_Q15 = 0;
+
     for( k = 0; k < 3; k++ ) {
         /* Safety margin for pitch gain control, to take into account factors
            such as state rescaling/rewhitening. */
```

tocklime, 2020-10-13:
yes, i can across that, and fixed it in the same way you got to. I think it compiled after that, but didn't run on my board well (I was using a WROOM not a WROVER). I tried disabling the PSRAM, and I think that got one step further, but I think I stopped around then. I'd like to get it working with a generic ESP32 (if it has to be wrover, then that's ok), and a basic I2S interface. I've got a lyrat mini on order, to play with that too, but I've got esp32s and I2S sound boards lying around now...

jorgenkraghjakobsen, 2020-10-14:
Hi Tocklime 
Non PSRAM works with limitations to meet snapcast min buffer size (400ms). 
ESP32-S0 is untested land - but should be added if it comes with PSRAM.  
/J