jorgenkraghjakobsen/snapclient #12: document required environment and working versions of adf and idf

tocklime, 2020-10-17:
Could you say anything about which versions or git revisions of esp-adf and esp-idf work, and where you expect them to be, and what environment variables should be set?

I've got
```
IDF_PYTHON_ENV_PATH=/home/greg/.espressif/python_env/idf4.3_py3.8_env
IDF_PATH=/home/greg/Documents/code/esp-idf
ADF_PATH=/home/greg/Documents/code/esp-adf
IDF_TOOLS_EXPORT_CMD=/home/greg/Documents/code/esp-idf/export.sh
IDF_TOOLS_INSTALL_CMD=/home/greg/Documents/code/esp-idf/install.sh
```
and esp-adf and esp-idf checkouts of current masters in sibling directories (as set in those environment variables), at gitrefs:
```
esp-idf: 0289d1cc81c210b719f28c65f113c45f9afd2c7b refs/heads/master
esp-adf: 789998a60d90ac3bba99d0dc869c72ff733d3c5a refs/heads/master
```
I can run idf.py menuconfig OK, but build fails in the linker step towards the end with 

```
/home/greg/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/audio_hal/libaudio_hal.a(es8388.c.obj):(.literal.es_read_reg+0x4): undefined reference to `i2c_bus_read_bytes'
/home/greg/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/audio_hal/libaudio_hal.a(es8388.c.obj):(.literal.es_write_reg+0x0): undefined reference to `i2c_bus_write_bytes'
/home/greg/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/audio_hal/libaudio_hal.a(es8388.c.obj):(.literal.es8388_deinit+0x0): undefined reference to `i2c_bus_delete'
/home/greg/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/audio_hal/libaudio_hal.a(es8388.c.obj):(.literal.i2c_init+0x4): undefined reference to `i2c_bus_create'
/home/greg/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/audio_hal/libaudio_hal.a(es8388.c.obj): in function `es_read_reg':
/home/greg/Documents/code/esp-adf/components/audio_hal/driver/es8388/es8388.c:63: undefined reference to `i2c_bus_read_bytes'
/home/greg/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/audio_hal/libaudio_hal.a(es8388.c.obj): in function `es_write_reg':
/home/greg/Documents/code/esp-adf/components/audio_hal/driver/es8388/es8388.c:58: undefined reference to `i2c_bus_write_bytes'
/home/greg/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/audio_hal/libaudio_hal.a(es8388.c.obj): in function `es8388_deinit':
/home/greg/Documents/code/esp-adf/components/audio_hal/driver/es8388/es8388.c:235: undefined reference to `i2c_bus_delete'
/home/greg/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/audio_hal/libaudio_hal.a(es8388.c.obj): in function `i2c_init':
/home/greg/Documents/code/esp-adf/components/audio_hal/driver/es8388/es8388.c:76: undefined reference to `i2c_bus_create'
```

I tried with the version of idf that is a submodule of adf too, but that gives different errors much earlier in the build process:

```
-- Detecting CXX compile features - done
CMake Error at /home/greg/Documents/code/snapclient/main/CMakeLists.txt:1 (idf_component_register):
  Unknown CMake command "idf_component_register".
```

jorgenkraghjakobsen, 2020-10-17:
What Lyrat board are you on - i have only tested V4.3 and custom audio 
/j 

8bc19ba893e5544d571a753d82b44a84799b94b1 refs/heads/master
1c0ef1e3cf569127e8758fa57acf330987f2c17e refs/heads/master
akobsen@t460s:~/esp/snapclient$ echo $IDF_PYTHON_ENV_PATH 
/home/jakobsen/.espressif/python_env/idf4.1_py3.8_env
jakobsen@t460s:~/esp/snapclient$ echo $IDF_PATH 
/home/jakobsen/esp/esp-idf
jakobsen@t460s:~/esp/snapclient$ echo $ADF_PATH 
/home/jakobsen/esp/esp-adf
jakobsen@t460s:~/esp/snapclient$ echo $IDF_TOOLS_EXPORT_CMD
/home/jakobsen/esp/esp-idf/export.sh
jakobsen@t460s:~/esp/snapclient$ echo $IDF_TOOLS_INSTALL_CMD
/home/jakobsen/esp/esp-idf/install.sh

jorgenkraghjakobsen, 2020-10-17:
I am cleaning up code moving unrelated stuf to subcomponent - my code is not stable right now. /j

tocklime, 2020-10-18:
I'm not even getting to 'what board do i want to use' (wroom with a custom i2s dac, so on snapclient i just want to configure pins for bck, lrck and data).

I've updated esp-idf and adf to the refs you mentioned (+recusive submodule updates), got a clean checkout of this repo, made the fix to quant_LTP_gains.c mentioned here: https://github.com/jorgenkraghjakobsen/snapclient/issues/11#issuecomment-707351159, and then run idf.py build, and it fails at the linker step with these errors.

I imagine there's some library, tool or pip package that should be installed but isn't. Or something else wrong in my environment...

jorgenkraghjakobsen, 2020-10-18:
Can you run idf.py menuconfig 
under Audio HAL - select custom 
under ESP32 audio buffer and I2S pin config - uncheck PSRAM - set I2S pins 
/j

tocklime, 2020-10-18:
Thanks for your help.

setting HAL - custom makes it missing a board.h - presumably I'm supposed to supply that.
```
/home/greg/Documents/code/esp-adf/components/display_service/led_bar/led_bar_ws2812.c:27:10: fatal error: board.h: No such file or directory
 #include "board.h"
          ^~~~~~~~~
```
Obviously I'll need to make these changes before I can flash to my setup, but I haven't yet managed to build snapclient for any configuration of hardware.

tocklime, 2020-10-18:
huh, this is weird. Just tried again on my windows install and got it compiling. now getting ip connect problems to my snapcast server, but I guess it's progress. No idea what's wrong with my linux setup though.

tocklime, 2020-10-18:
aha. setting port manually needs htons(1704) not just 1704. (my snapcast server is running as a home assistant add on, and mdns isn't working, so have hard coded it. I got sound out! with no volume control and a fair amount of crackling (presumably buffer over/underruns).