jorgenkraghjakobsen/snapclient #22: Unable to compile

jirmjos, 2021-02-27:
Hi Jorgen,

First all, many thanks to share this really interesting project.

I trying to compile from your source but I´m unable to build because Cmake raise some errors.
I setup esp-idf and isp-adf from scratch and can build any example project with no issues, but your project seems have missing some link to the components or something.

I can make menuconfig without issues and change de audio board target to one test cheap 5102A dac board (I´m in process to buy a Merus amp with MA12070P chip)  but error are raised with any audio board type.
I post the error I have with 5102 dac board:
`[1078/1274] Building C object esp-idf/wifi_interface/CMakeFiles/__idf_wifi_interface.dir/wifi_interface.c.obj                                                                                                                                FAILED: esp-idf/wifi_interface/CMakeFiles/__idf_wifi_interface.dir/wifi_interface.c.obj                                                                                                                                                      ccache C:\Users\josei\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe -DADF_VER=\"1ece283-dirty\" -Iconfig -I../components/wifi_interface/include -I../../components/newlib/platform_include -I../../components/freertos/include -I../../components/freertos/xtensa/include -I../../components/heap/include -I../../components/log/include -I../../components/lwip/include/apps -I../../components/lwip/include/apps/sntp -I../../components/lwip/lwip/src/include -I../../components/lwip/port/esp32/include -I../../components/lwip/port/esp32/include/arch -I../../components/soc/src/esp32/. -I../../components/soc/src/esp32/include -I../../components/soc/include -I../../components/esp_rom/include -I../../components/esp_common/include -I../../components/esp_system/include -I../../components/xtensa/include -I../../components/xtensa/esp32/include -I../../components/esp32/include -I../../components/driver/include -I../../components/driver/esp32/include -I../../components/esp_ringbuf/include -I../../components/efuse/include -I../../components/efuse/esp32/include -I../../components/espcoredump/include -I../../components/esp_timer/include -I../../components/esp_ipc/include -I../../components/soc/soc/esp32/include -I../../components/soc/soc/esp32/../include -I../../components/soc/soc/esp32/private_include -I../../components/vfs/include -I../../components/esp_wifi/include -I../../components/esp_wifi/esp32/include -I../../components/esp_event/include -I../../components/esp_netif/include -I../../components/esp_eth/include -I../../components/tcpip_adapter/include -I../../components/app_trace/include -mlongcalls -Wno-frame-address   -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -mfix-esp32-psram-cache-issue -mfix-esp32-psram-cache-strategy=memw -Og -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v4.2-dirty\" -DESP_PLATFORM -MD -MT esp-idf/wifi_interface/CMakeFiles/__idf_wifi_interface.dir/wifi_interface.c.obj -MF esp-idf\wifi_interface\CMakeFiles\__idf_wifi_interface.dir\wifi_interface.c.obj.d -o esp-idf/wifi_interface/CMakeFiles/__idf_wifi_interface.dir/wifi_interface.c.obj   -c ../components/wifi_interface/wifi_interface.c                                                                                                                                                                                                         In file included from ../components/wifi_interface/wifi_interface.c:8:                                                                                                                                                                       ../components/wifi_interface/include/wifi_interface.h:6:1: error: unknown type name 'EventGroupHandle_t'; did you mean 'SemaphoreHandle_t'?                                                                                                   EventGroupHandle_t s_wifi_event_group;                                                                                                                                                                                                       ^~~~~~~~~~~~~~~~~~                                                                                                                                                                                                                           SemaphoreHandle_t                                                                                                                                                                                                                           ../components/wifi_interface/wifi_interface.c: In function 'event_handler':                                                                                                                                                                  ../components/wifi_interface/wifi_interface.c:42:26: warning: passing argument 1 of 'xEventGroupSetBits' makes pointer from integer without a cast [-Wint-conversion]                                                                               xEventGroupSetBits(s_wifi_event_group, WIFI_FAIL_BIT);                                                                                                                                                                                                          ^~~~~~~~~~~~~~~~~~                                                                                                                                                                                                 In file included from ../components/wifi_interface/wifi_interface.c:14:                                                                                                                                                                      ../../components/freertos/include/freertos/event_groups.h:483:52: note: expected 'EventGroupHandle_t' {aka 'void *'} but argument is of type 'int'                                                                                            EventBits_t xEventGroupSetBits( EventGroupHandle_t xEventGroup, const EventBits_t uxBitsToSet ) PRIVILEGED_FUNCTION;                                                                                                                                                         ~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~                                                                                                                                                                              ../components/wifi_interface/wifi_interface.c:49:24: warning: passing argument 1 of 'xEventGroupSetBits' makes pointer from integer without a cast [-Wint-conversion]                                                                             xEventGroupSetBits(s_wifi_event_group, WIFI_CONNECTED_BIT);                                                                                                                                                                                                     ^~~~~~~~~~~~~~~~~~                                                                                                                                                                                                   In file included from ../components/wifi_interface/wifi_interface.c:14:                                                                                                                                                                      ../../components/freertos/include/freertos/event_groups.h:483:52: note: expected 'EventGroupHandle_t' {aka 'void *'} but argument is of type 'int'                                                                                            EventBits_t xEventGroupSetBits( EventGroupHandle_t xEventGroup, const EventBits_t uxBitsToSet ) PRIVILEGED_FUNCTION;                                                                                                                                                         ~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~                                                                                                                                                                              ../components/wifi_interface/wifi_interface.c: In function 'wifi_init_sta':         [... cut ...] es/__idf_opus.dir/opus/src/analysis.c.obj                                                                                                                                                 ninja: build stopped: subcommand failed.                                                                                                                                                                                                     ninja failed with exit code 1        `

Any help will be appreciate.

Regards

jorgenkraghjakobsen, 2021-02-27:
Hi Jose 
...
components/wifi_interface/include/wifi_interface.h:6:1: error: unknown type name 'EventGroupHandle_t'; did you mean 'SemaphoreHandle_t'? 
Compilation stopped here. 

I fixed this one move 
#include "wifi_interface.h"

below the other includes 
/j

jirmjos, 2021-02-28:
Hi Jorgen,
Many thanks for your reply

Wifi issue seems fixed for compilation. 
Now still error about opus component  that seems older one because is referenced in other issue 
 https://github.com/jorgenkraghjakobsen/snapclient/issues/11 .
`FAILED: esp-idf/opus/CMakeFiles/__idf_opus.dir/opus/silk/quant_LTP_gains.c.obj                                                                                              ccache C:\Users\josei\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe -DADF_VER=\"6a52d87-dirty\" -DHAVE_CONFIG_H -Iconfig -I../components/opus -I../components/opus/opus/include -I../components/opus/opus/silk -I../components/opus/opus/silk/fixed -I../components/opus/opus/silk/float -I../components/opus/opus/celt -I../../components/newlib/platform_include -I../../components/freertos/include -I../../components/freertos/xtensa/include -I../../components/heap/include -I../../components/log/include -I../../components/lwip/include/apps -I../../components/lwip/include/apps/sntp -I../../components/lwip/lwip/src/include -I../../components/lwip/port/esp32/include -I../../components/lwip/port/esp32/include/arch -I../../components/soc/src/esp32/. -I../../components/soc/src/esp32/include -I../../components/soc/include -I../../components/esp_rom/include -I../../components/esp_common/include -I../../components/esp_system/include -I../../components/xtensa/include -I../../components/xtensa/esp32/include -I../../components/esp32/include -I../../components/driver/include -I../../components/driver/esp32/include -I../../components/esp_ringbuf/include -I../../components/efuse/include -I../../components/efuse/esp32/include -I../../components/espcoredump/include -I../../components/esp_timer/include -I../../components/esp_ipc/include -I../../components/soc/soc/esp32/include -I../../components/soc/soc/esp32/../include -I../../components/soc/soc/esp32/private_include -I../../components/vfs/include -I../../components/esp_wifi/include -I../../components/esp_wifi/esp32/include -I../../components/esp_event/include -I../../components/esp_netif/include -I../../components/esp_eth/include -I../../components/tcpip_adapter/include -I../../components/app_trace/include -mlongcalls -Wno-frame-address   -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -mfix-esp32-psram-cache-issue -mfix-esp32-psram-cache-strategy=memw -Og -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v4.2-dirty\" -DESP_PLATFORM -MD -MT esp-idf/opus/CMakeFiles/__idf_opus.dir/opus/silk/quant_LTP_gains.c.obj -MF esp-idf\opus\CMakeFiles\__idf_opus.dir\opus\silk\quant_LTP_gains.c.obj.d -o esp-idf/opus/CMakeFiles/__idf_opus.dir/opus/silk/quant_LTP_gains.c.obj   -c ../components/opus/opus/silk/quant_LTP_gains.c                                ../components/opus/opus/silk/quant_LTP_gains.c: In function 'silk_quant_LTP_gains':                                                                                         ../components/opus/opus/silk/quant_LTP_gains.c:127:21: error: 'res_nrg_Q15' may be used uninitialized in this function [-Werror=maybe-uninitialized]                                 res_nrg_Q15 = silk_RSHIFT32( res_nrg_Q15, 2 );                                                                                                                     cc1.exe: some warnings being treated as errors                                                                                                                              [1160/1281] Building C object esp-idf/opus/CMakeFiles/__idf_opus.dir/opus/silk/resampler_private_IIR_FIR.c.obj                                                              ninja: build stopped: subcommand failed.                                                                                                                                    ninja failed with exit code 1`

I but i but i was able to fix it modifing file  ...\opus\silk\quant_LTP_gains.c, trough  adding res_nrg_Q15 = 0; as per related.
I think will be fine you modiffy this on your sources too.

Now I waw able to compile, with only a minor warnings:

`Generated C:/Users/josei/esp/snapclient/build/bootloader/bootloader.bin                                                                                                     [1204/1281] Building C object esp-idf/websocket/CMakeFiles/__idf_websocket.dir/websocket_server.c.obj                                                                       ../components/websocket/websocket_server.c: In function 'handle_read':                                                                                                      ../components/websocket/websocket_server.c:51:12: warning: 'return' with a value, in function returning void                                                                     return NULL;                                                                                                                                                                       ^~~~                                                                                                                                                            ../components/websocket/websocket_server.c:42:13: note: declared here                                                                                                 [... cut ...]                           [1281/1281] Generating binary image from built executable                                                                                                                   esptool.py v3.0                                                                                                                                                             Generated C:/Users/josei/esp/snapclient/build/snapclient.bin `

Now finally I can starting test on hardware !!!

Best regards
Jose

jirmjos, 2021-03-01:
Resolved

closed 2021-03-01 by jirmjos