jorgenkraghjakobsen/snapclient #23: unable to boot on esp32 wroom module board

jirmjos, 2021-02-28:
Hi Jorgen,

Sorry to bother you, so much.

Flashed board (at this moment I´m only have wroom modules at hand) but now it falls to a looping reboot I think is because try to init psram that this ESP32 module don´t have.
I try to disable psram trough idf.py menuconfig, but seems no effect.

I don´t know how to fix it.

Thats the monitor output:
`Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7036
load:0x40078000,len:13772
ho 0 tail 12 room 4
load:0x40080400,len:4632
0x40080400: _init at ??:?

entry 0x400806f4
I (30) boot: ESP-IDF v4.2-dirty 2nd stage bootloader
I (31) boot: compile time 09:36:25
I (31) boot: chip revision: 1
I (34) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (41) boot.esp32: SPI Speed      : 40MHz
I (46) boot.esp32: SPI Mode       : DIO
I (50) boot.esp32: SPI Flash Size : 4MB
I (55) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (79) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (86) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (94) boot:  3 factory          factory app      00 00 00010000 00100000
I (101) boot:  4 ota_0            OTA app          00 10 00110000 00100000
I (109) boot:  5 ota_1            OTA app          00 11 00210000 00100000
I (116) boot: End of partition table
I (120) boot: Defaulting to factory image
I (125) boot_comm: chip revision: 1, min. application chip revision: 0
I (132) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x26618 (157208) map
I (207) esp_image: segment 1: paddr=0x00036640 vaddr=0x3ffb0000 size=0x03928 ( 14632) load
I (214) esp_image: segment 2: paddr=0x00039f70 vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at C:/Users/josei/esp/components/freertos/xtensa/xtensa_vectors.S:1730

I (215) esp_image: segment 3: paddr=0x0003a37c vaddr=0x40080404 size=0x05c9c ( 23708) load
I (234) esp_image: segment 4: paddr=0x00040020 vaddr=0x400d0020 size=0xa85bc (689596) map
0x400d0020: _stext at ??:?

I (522) esp_image: segment 5: paddr=0x000e85e4 vaddr=0x400860a0 size=0x154f8 ( 87288) load
0x400860a0: esp_flash_erase_region at C:/Users/josei/esp/components/spi_flash/esp_flash_api.c:331

I (579) boot: Loaded app from partition at offset 0x10000
I (579) boot: Disabling RNG early entropy source...
I (580) psram: This chip is ESP32-D0WD
E (585) psram: PSRAM ID read error: 0xffffffff
E (589) cpu_start: Failed to init external RAM!
Re-enable cpu cache.

abort() was called at PC 0x40081ec6 on core 0
0x40081ec6: call_start_cpu0 at C:/Users/josei/esp/components/esp32/cpu_start.c:191 (discriminator 1)

Backtrace:0x4008c912:0x3ffe3b90 0x4008cfd1:0x3ffe3bb0 0x40093e4e:0x3ffe3bd0 0x40081ec6:0x3ffe3c40 0x400790e2:0x3ffe3c80 |<-CORRUPTED
0x4008c912: panic_abort at C:/Users/josei/esp/components/esp_system/panic.c:330

0x4008cfd1: esp_system_abort at C:/Users/josei/esp/components/esp_system/system_api.c:106

0x40093e4e: abort at C:/Users/josei/esp/components/newlib/abort.c:46

0x40081ec6: call_start_cpu0 at C:/Users/josei/esp/components/esp32/cpu_start.c:191 (discriminator 1)

ELF file SHA256: e70d25980491c421`

Regards,
Jose,

jorgenkraghjakobsen, 2021-02-28:
Hi Jose 
No problem - I will see if I can get it running on a non PSRAM board later today.
/j

jirmjos, 2021-02-28:
Hi Again,

I get my wroom module running.

Easy anougth fix it, because only needed disable esternal spi ram trough menuconfig: Component config ---> ESP32-specific ---> Support for external, SPI-connected RAM

Fast reading errors I messed up, PSRAM with SPI RAM.

Now seems my module are up, and next step tunning configuration to my dac board and propper attach it to esp32 (I see some references to ma12070 and i2c bus (that it´s not propper for my dac board) and finally I think can last configure and test snapcast.

Monitor output from boot:

`I (29) boot: ESP-IDF v4.2-dirty 2nd stage bootloader                                                                    I (29) boot: compile time 18:46:05                                                                                      I (29) boot: chip revision: 1                                                                                           I (32) boot_comm: chip revision: 1, min. bootloader chip revision: 0                                                    I (39) boot.esp32: SPI Speed      : 40MHz                                                                               I (44) boot.esp32: SPI Mode       : DIO                                                                                 I (48) boot.esp32: SPI Flash Size : 4MB                                                                                 I (53) boot: Enabling RNG early entropy source...                                                                       I (58) boot: Partition Table:                                                                                           I (62) boot: ## Label            Usage          Type ST Offset   Length                                                 I (69) boot:  0 nvs              WiFi data        01 02 00009000 00004000                                               I (76) boot:  1 otadata          OTA data         01 00 0000d000 00002000                                               I (84) boot:  2 phy_init         RF data          01 01 0000f000 00001000                                               I (91) boot:  3 factory          factory app      00 00 00010000 00100000                                               I (99) boot:  4 ota_0            OTA app          00 10 00110000 00100000                                               I (106) boot:  5 ota_1            OTA app          00 11 00210000 00100000                                              I (114) boot: End of partition table                                                                                    I (118) boot: Defaulting to factory image                                                                               I (123) boot_comm: chip revision: 1, min. application chip revision: 0                                                  I (130) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x21d3c (138556) map                               I (192) esp_image: segment 1: paddr=0x00031d64 vaddr=0x3ffb0000 size=0x038b4 ( 14516) load                              I (198) esp_image: segment 2: paddr=0x00035620 vaddr=0x40080000 size=0x00404 (  1028) load                              0x40080000: _WindowOverflow4 at C:/Users/josei/esp/components/freertos/xtensa/xtensa_vectors.S:1730                                                                                                                                             I (199) esp_image: segment 3: paddr=0x00035a2c vaddr=0x40080404 size=0x0a5ec ( 42476) load                              I (226) esp_image: segment 4: paddr=0x00040020 vaddr=0x400d0020 size=0x9eb38 (650040) map                               0x400d0020: _stext at ??:?                                                                                                                                                                                                                      I (474) esp_image: segment 5: paddr=0x000deb60 vaddr=0x4008a9f0 size=0x0c568 ( 50536) load                              0x4008a9f0: vTaskSuspend at C:/Users/josei/esp/components/freertos/tasks.c:1838                                                                                                                                                                 I (509) boot: Loaded app from partition at offset 0x10000                                                               I (509) boot: Disabling RNG early entropy source...                                                                     I (509) cpu_start: Pro cpu up.                                                                                          I (513) cpu_start: Application information:                                                                             I (518) cpu_start: Project name:     snapclient                                                                         I (523) cpu_start: App version:      6a52d87-dirty                                                                      I (528) cpu_start: Compile time:     Feb 28 2021 18:45:47                                                               I (534) cpu_start: ELF file SHA256:  3cc6ce130b85b11e...                                                                I (540) cpu_start: ESP-IDF:          v4.2-dirty                                                                         I (546) cpu_start: Starting app cpu, entry point is 0x400819fc                                                          0x400819fc: call_start_cpu1 at [... cut ...]                                                                           I (4092) I2S: Wait: no data in buffer 5 0                                                                               I (4492) I2S: Wait: no data in buffer 6 0                                                                               I (4892) I2S: Wait: no data in buffer 7 0                                                                               I (5292) I2S: Wait: no data in buffer 8 0`

First step seems done:

jorgenkraghjakobsen, 2021-03-01:
Hi Jose - Nice yes you need to setup you own custom board - or get one of the MA12070p based setup. /J

jorgenkraghjakobsen, 2021-03-01:
The repo is very much work in progress. The MA120 that I am using is not released yet. /J

jirmjos, 2021-03-01:
Hi Jorgen,

I´m in process to buy this MA12070P board https://www.audiophonics.fr/en/amplifier-boards/balanced-class-d-10-20-21-40-amplifier-module-ma12070-160w-4-ohm-p-14732.html.
Hopping they build enough well the board and mantain the good MA12070p spechs chips and his performance.
In few days I hoppe have this Merus board in my hands and too some purchased ESP32 Wroover boards to start real test with target final hardware.

Meanwhile I try to steep forward to test and understand your snapcast client to tunning but with the hardware I have now .

Regards