jorgenkraghjakobsen/snapclient #26: Periodic crashes / reboots when playing

kabbi, 2021-03-29:
I've updated your code to support bare PCM5102 (without i2c control, just mocked all functions from pcm51xx to do nothing), and I'm using esp32 d1 mini (no psram) module.

Audio is playing fine, but after 3 to 10 seconds it automatically reboots.

```
I (17969) TAVG: Time input :     -352394.729538
I (17969) TAVG: Time avg :               730812  0 
I (17969) SNAPCAST: Tclientdif :    -352394.729 
I (17969) SNAPCAST: Tclientdif : return value 729538 
I (17979) SNAPCAST: TM loopback latency: 14.0 ms
I (18109) I2S: Error readding audio from ring buf 1012 3528 
I (18119) I2S: Error readding audio from ring buf 2504 1015093583 
I (18299) I2S: Error readding audio from ring buf 1920 3528 
I (18299) I2S: Error readding audio from ring buf 1596 560073636 
I (18479) I2S: Error readding audio from ring buf 2828 3528 
I (18479) I2S: Error readding audio from ring buf 688 -267781335 
I (18729) I2S: Error readding audio from ring buf 196 3528 
I (18729) I2S: Error readding audio from ring buf 10400 -425862273 
E (18739) I2S: i2s_write_expand(1115): size is too large
.../esp-idf/components/esp_ringbuf/ringbuf.c:648 (prvReturnItemByteBuf)- assert failed!

abort() was called at PC 0x40087f07 on core 0
0x40087f07: prvReturnItemByteBuf at .../esp-idf/components/esp_ringbuf/ringbuf.c:648 (discriminator 1)

Backtrace:0x40088e73:0x3fff33c0 0x40089645:0x3fff33e0 0x4009058e:0x3fff3400 0x40087f07:0x3fff3470 0x40088bcd:0x3fff3490 0x400e1271:0x3fff34b0 0x4008cc61:0x3fffc530
0x40088e73: panic_abort at .../esp-idf/components/esp_system/panic.c:356

0x40089645: esp_system_abort at ...esp-idf/components/esp_system/system_api.c:112

0x4009058e: abort at .../esp-idf/components/newlib/abort.c:46

0x40087f07: prvReturnItemByteBuf at .../esp-idf/components/esp_ringbuf/ringbuf.c:648 (discriminator 1)

0x40088bcd: vRingbufferReturnItem at .../esp-idf/components/esp_ringbuf/ringbuf.c:1268 (discriminator 2)

0x400e1271: dsp_i2s_task_handler at .../snapclient/build/../components/dsp_processor/dsp_processor.c:493

0x4008cc61: vPortTaskWrapper at .../esp-idf/components/freertos/port/xtensa/port.c:168

```

kabbi, 2021-03-29:
Anyway, it works fine for ESP32-CAM board with psram! Thank you for this great project!

jorgenkraghjakobsen, 2021-03-30:
Hi Dmitry - Good to hear - I am warming up to do a larger refactoring - main motivation will be to end up with a snapclient component that is easy to reuse in other projects. /J