jorgenkraghjakobsen/snapclient #31: Hardware questions

LeoSum8, 2021-06-11:
Hi there,
I am looking into building this project to replace my rPi and openWRT based clients which annoy me with either too long boot time or too high power consumption. As covered in #28, none of this seems to be an issue here :).

As there is no "beginners guide" here, I have to ask some basic questions to get myself started. 
I'd be happy to contribute a writeup of my journey, if this is something you want to include in the documentation of this project.

So for starters, as far as I understand, hardware-wise I'll need:

| Component  | my choice |
| ------------- | ------------- |
| ESP32 Board  | TTGO T8 WROVER  |
| AMP  | not sure yet, something based on MA12070P  |
| Speakers  | JBL Control pair  |
| Power Supply | not sure yet, will depend on AMP  |

So the component I am having trouble selecting is the amp really. Would this one work? https://www.ebay.de/itm/124534259519

It seems to be quite hard to find something similar that ships from within the EU.

Thanks in advance for any pointers!
Leo

jorgenkraghjakobsen, 2021-06-12:
https://www.ebay.de/itm/114590460736?hash=item1aae1fab40:g:C4gAAOSwB65f4vW4
Make sure your get the digital input (ma12070p) device. You can supply up to 18-24V will be more then enough for your JBL control speakers. 
The repo as is now is a bit split between the syncing implementation/strategy - plan to clean up during next week. 
Where are you located in EU? 
Regards Jørgen

LeoSum8, 2021-06-14:
Thanks for your answer!
The one you linked seems to miss the trailing "P" just like the one I found. Also the description doesn't say anything about digital inputs and the pictures of the board don't show the Pins for it. Compared to your Readme I am missing the I2S_BCK, I2S_WS, I2S_DI Pins. 

IIS is only mentioned in the title, so I am a bit sceptical.

I am located in Germany.

jorgenkraghjakobsen, 2021-06-14:
Hi Leo 
I found one of the China outlet where you can select the analog/digital input - but you will still suffer the 6-10 weeks lead time. 
Alternative you can get the amp direct from Infineon -  https://www.infineon.com/cms/en/product/evaluation-boards/ref_audio_d_ma12070p/ but it is quite pricy for a hobby project.
I am currently working on an ESP32/MA12070P design myself - It will be available in limited numbers around end next month. 
/j

jorgenkraghjakobsen, 2021-06-14:
Hi Leo - Drop me a mail on jkj@myrun.dk - I mite be able to help you out. /J

LeoSum8, 2021-06-14:
Sounds interesting, you've got mail :)

idpnd, 2021-06-26:
Hi Jorgen, this is the forum I gather? 

Thanks v much for your work on this, this is my first sub raspberry size project, so not much to expect from me in coding contributions. I've also taken the liberty of writing you an email, interested in this amp. I like the frugal consumption in standby mode, sort of lazy man's solution to the power on/off relais for remote applications ;)

What speaker(s) do you recommend? I'd consider one mono speaker per amp, since, as you like to say "getting the party started" wants a bit of power does it not.

Best regards

gschanuel, 2021-11-04:
Hello there
Would this board work? 
https://a.aliexpress.com/_mMKpWsS

jorgenkraghjakobsen, 2021-11-04:
Yes - It is one of the base Espressif Lyra dev board good for audio demo - not to rock the house. /J

LeoSum8, 2021-11-04:
Thanks!
For house-rocking-purposes I am still trying to find a source for MA12070P that is not in china ...