jorgenkraghjakobsen/snapclient #35: Building with just ESP-IDF

oresk, 2021-11-23:
First of all, thnx for all the work on this project!

Wanted to check one thing. You say in the readme that `The codebase is split into components and build on vanilla ESP-IDF.` 
I have just the esp-idf installed on my machine. When I try to start the menuconfig I get this error 
`CMake Error at CMakeLists.txt:5 (include):
  include could not find requested file:

    /CMakeLists.txt`

Which I think is produced by root CMakeLists.txt line `include($ENV{ADF_PATH}/CMakeLists.txt)`. 

What should I do with that? 
Install ADF, point the ADF-PATH to the IDF_PATH, or do some other change?

---- 
I tried just changing the ADF-PATH to IDF-PATH, and I get this error later in the menuconfig start up

`CMake Error at /home/lovro/esp/esp-idf/CMakeLists.txt:215 (idf_build_set_property):
  Unknown CMake command "idf_build_set_property".
Call Stack (most recent call first):
  CMakeLists.txt:5 (include)
`
My ESP-IDF version is: ESP-IDF v5.0-dev-329-g4a011f3183

Thnx in advance!

kiesel, 2021-11-28:
Stumbled over this problem as well; fixed by installing esp-adf and pointing ADF_PATH to that installation - following the steps as documented here: https://docs.espressif.com/projects/esp-adf/en/latest/get-started/#get-started-get-esp-adf

One could improve the documentation with that information, I guess:-)

oresk, 2021-12-06:
Thnx for the info. 
What ESP_IDF version are you using?

kiesel, 2021-12-06:
```
$ idf.py --version
ESP-IDF v4.3.1-dirty
```

oresk, 2021-12-07:
A bit of a context before I close the issue.

In the end I managed to build it by using 
current master (06.12.2021) of the adf 
and using the esp-idf 4.2.1 version.

All in all it was a bit of a hit and miss. Using the esp-idf from the adf did not work, and using the current 5.0 dev version of the esp-idf did not work also. 

Had to make a small change in the opus component code (initialize a variable on the start) so that the compiler does not complain. 

The esp connects to the snapcast server that I am running, but it crashes afterwards trying to connect to the pcm5102A (I think I actually have pcm5100 that the aliexpress people have sanded the top of and lasered on a new inscription, but it should work for testing just fine). 
The pcm51xx code in the repo has a prediction that it can control the volume by using the i2c, which does not happen on my board, so next step is to figure out that bit.

closed 2021-12-07 by oresk