jorgenkraghjakobsen/snapclient #39: Hardware Adaptions

Gerhardt-Tooling, 2022-01-08:
Hello together,

i found finally time to continue the work on my amplifier. I'm more the HW guy an struggle currently to compile the package in VS Code... Anyway i need then help to adapt this SW to my board. I would appreciate any support.
The HW description and Pinning of ESP WROOM Board is here: https://gerhardt-tooling.de/en/projects-2/multiroomaudio/
So far everything should be supportet by the ESP ADF.
I would need:
- a selection between Wired Lan with driver LAN8720A or W-Lan
- Audio DAC ES8388
- Digital Enable signal for the Amplifier

Thanks in Advance 
Kind Regards Martin

jorgenkraghjakobsen, 2022-01-09:
Hi Martin - Cool - you will be limited to 250ms network audio buffer on the client side due to limited memory on the wroom board. But on the other hand your will have LAN and will not suffer from the network jitter/lost packages as on WLAN setup. 
I do not have any experience on the TAS amps or the DAC you are using. But there should be a DAC/Amp abstraction layer in the code that you need to adapt. 
The switch between WLAN and LAN must be added to the system config in kconfig style. 
Happy coding :-)

CarlosDerSeher, 2022-03-24:
Lyrat Board v4.3 is using es8388 isn't it. You could use its HAL to start with. I am working on a similar project but didn't have time to finish the PCB yet. You could checkout my fork, it's a bit messy though it should work using Lyrat flawlessly. Using NETCONN branch you should be able to use a network buffer of <=500ms with WROOM modules. LAN isn't implemented yet but there are examples using LAN8720 you could start with.