jorgenkraghjakobsen/snapclient #48: Transferring ownership of the repository

CarlosDerSeher, 2023-12-14:
It seems you are not active on this anymore. As my fork is pretty mature now I was thinking, maybe you could transfer to project to me?

Thanks for all your work and greetings from Austria
Karl

jorgenkraghjakobsen, 2023-12-15:
Hi Karl Thanks.I would be very happy to do so.And yes I have moved on, started my on consultancy shop that is planned to grow in to some thing bigger.All my time is now used in open source chip design tools and PDKs.Heading off to Austria next Friday to spend the xmes in the snow.Jørgen14. dec.

TekniskSupport, 2023-12-15:
A BIG -Thank you- for your time spent and work on this so far to make it what it is @jorgenkraghjakobsen

CarlosDerSeher, 2023-12-15:
Thank you and good luck and much success on your way :)

You go skiing to the mountains? Tyrol .. Salzburg? I am from the flatlands, near Vienna on the countryside. Highest mountain here is maybe around 500m XD

CarlosDerSeher, 2024-01-09:
@jorgenkraghjakobsen hey I hope you had a good vacation?
Just a little reminder, if you find some time to do this it would be really great :)

CarlosDerSeher, 2024-02-06:
@jorgenkraghjakobsen Not sure if this really is what you want: https://docs.github.com/en/repositories/creating-and-managing-repositories/transferring-a-repository

if not, you could just link to my fork from Readme's first line with the hint that future deleopment is happening there.

jagjordi, 2024-10-12:
Just a reminder about this :) I have started to use @CarlosDerSeher and it works quite well. Still when you google ESP32 snapclient @jorgenkraghjakobsen repository comes up. It would be nice to unify both of them in some way while still maintaining credit to both of you :)

CarlosDerSeher, 2024-10-12:
@jagjordi i guess you'll have to mention @jorgenkraghjakobsen directly 😉

miLORD1337, 2025-03-20:
Hey @jorgenkraghjakobsen, 
it'd be really nice if you could do this transfer of ownership. Maybe it got pushed back and forgotten?

Kind regards,
miLORD

miLORD1337, 2025-04-04:
Hey @CarlosDerSeher,
maybe leaving the fork network would be an option for your repo? It would be standalone then. Probably less hidden that way?
I don't know everything about the implications that has though.

Kind regards,
miLORD

CarlosDerSeher, 2025-04-04:
Could be an option. Didn't think about this probability. Thank you

CarlosDerSeher, 2025-04-04:
https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/working-with-forks/detaching-a-fork

This seems the way to go. Not sure which of the 3 routes will be best.

CarlosDerSeher, 2025-04-06:
I've just created a ticket with github support to "extract" the fork. This will make my fork a standalone repository and hopefully make it more visible.

> Created via Virtual Agent flow
Please detach/extract the fork carlosderseher/snapclient (CarlosDerSeher/snapclient) - this repository is public

Extract child forks to a new network
Customer justification: As the original developer seems to have abondond the project I became the new maintainer. The owner does not respond anymore and neither does he accept any pull requests

CarlosDerSeher, 2025-04-07:
@miLORD1337 All done. My fork has now it's own fork network.

```
Hi CarlosDerSeher,
 
Thanks for reaching out!
 
Kindly note that, I have extracted that fork for you. The repository and its child forks are now in a new network
 
I hope this helps!
 
Let me know if you have further concerns.
 
Regards,
 
Travis
GitHub Support
```

miLORD1337, 2025-04-07:
That's so nice! Well deserved 😄 
Now Google needs to catch up, but it will.